HKUST-Aerial-Robotics/A-LOAM #2: Segmentation fault

ymxlzgy, 2019-03-21:
Sorry to bother you, but when I launch the ascanRegistration from aloam_velodyne_64.launch, I obtained this error and I saw 
" #7  std::vector<pcl::PointXYZI, Eigen::aligned_allocator<pcl::PointXYZI> >::~vector (this=0x7fffffffc758, __in_chrg=<optimized out>)
 "
"#8  pcl::PointCloud<pcl::PointXYZI>::~PointCloud (this=0x7fffffffc720,  __in_chrg=<optimized out>)" in gdb, maybe something is wrong with eigen. I use eigen3.3.7 and pcl1.9, Can you help me ?

kxhit, 2019-03-22:
@ymxlzgy I successfully run the code by building ceres and pcl1.8 with eigen3.2.9. Hope it helps.

ymxlzgy, 2019-03-22:
@kxhit  You are such a clever boy!

SampsonM8, 2024-03-06:
Caused By -O3 compile optimization. 
Just Cancel -O3 optimization