HKUST-Aerial-Robotics/A-LOAM #3: Bad result of KITTI 00 sequence

langelesl, 2019-03-21:
Hi, thanks for sharing your code. I used [evo](https://github.com/MichaelGrupp/evo) to evaluate the trajectory I got from running your codes in ROS. But the result is not that ideal like you got in KITTI ranking. So there may be something wrong with my evaluation or steps. Could you please give me some advices on evaluating or tuning?

[image]

After running velodyne aloam_velodyne_64.launch I played rosbag of 00 sequence (2011_10_03_drive_0027) instead of kitti_helper.launch and odom data in folders. The running speed of bag was set to 0.1. And I remapped the published topic of point cloud to "/velodyne_points". The command "rostopic echo /laser_odom_to_unit/pose/pose > result.txt" is used to record the trajectory. 

Thanks.

qintonguav, 2019-03-31:
please use the topic "/aft_mapped_to_init"  for evaulation.

langelesl, 2019-04-04:
> please use the topic "/aft_mapped_to_init" for evaulation.

You are right. The trajectory looks better after map optimizaiton. Thanks.

closed 2019-04-04 by langelesl