HKUST-Aerial-Robotics/A-LOAM #5: Can this package successfully build on ubuntu 14.04(x64) ?

wykxwyc, 2019-04-23:
I build this A-LOAM on my ubuntu14.04(x64, virtual machine) , but it seems that the it's not compatiable with some library in ***ceres*** and ***Eigen***. 
I also build the package in Ubuntu16.04, and this problem didn't occur.
The following is the error information, could you please give me some advice?

```make
In file included from /usr/include/eigen3/Eigen/Geometry:40:0,
                 from /usr/include/pcl-1.7/pcl/point_cloud.h:47,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:41:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h: In instantiation of ‘Eigen::Quaternion<typename Eigen::internal::traits<T>::Scalar> Eigen::QuaternionBase<Derived>::slerp(const Scalar&, const Eigen::QuaternionBase<OtherDerived>&) const [with OtherDerived = Eigen::Quaternion<ceres::Jet<double, 7>, 0>; Derived = Eigen::Quaternion<ceres::Jet<double, 7>, 0>; typename Eigen::internal::traits<T>::Scalar = ceres::Jet<double, 7>; Eigen::QuaternionBase<Derived>::Scalar = ceres::Jet<double, 7>]’:
/home/either/catkin_tws/src/A-LOAM/src/lidarFactor.hpp:81:15:   required from ‘bool LidarPlaneFactor::operator()(const T*, const T*, T*) const [with T = ceres::Jet<double, 7>]’
/usr/local/include/ceres/internal/variadic_evaluate.h:168:26:   required from ‘static bool ceres::internal::VariadicEvaluate<Functor, T, N0, N1, 0, 0, 0, 0, 0, 0, 0, 0>::Call(const Functor&, const T* const*, T*) [with Functor = LidarPlaneFactor; T = ceres::Jet<double, 7>; int N0 = 4; int N1 = 3]’
/usr/local/include/ceres/internal/autodiff.h:290:45:   required from ‘static bool ceres::internal::AutoDiff<Functor, T, N0, N1, N2, N3, N4, N5, N6, N7, N8, N9>::Differentiate(const Functor&, const T* const*, int, T*, T**) [with Functor = LidarPlaneFactor; T = double; int N0 = 4; int N1 = 3; int N2 = 0; int N3 = 0; int N4 = 0; int N5 = 0; int N6 = 0; int N7 = 0; int N8 = 0; int N9 = 0]’
/usr/local/include/ceres/autodiff_cost_function.h:218:25:   required from ‘bool ceres::AutoDiffCostFunction<CostFunctor, kNumResiduals, N0, N1, N2, N3, N4, N5, N6, N7, N8, N9>::Evaluate(const double* const*, double*, double**) const [with CostFunctor = LidarPlaneFactor; int kNumResiduals = 1; int N0 = 4; int N1 = 3; int N2 = 0; int N3 = 0; int N4 = 0; int N5 = 0; int N6 = 0; int N7 = 0; int N8 = 0; int N9 = 0]’
/home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:602:1:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:712:7: error: no match for ‘operator<’ (operand types are ‘Eigen::QuaternionBase<Eigen::Quaternion<ceres::Jet<double, 7>, 0> >::Scalar {aka ceres::Jet<double, 7>}’ and ‘int’)
   if(d<0) scale1 = -scale1;
       ^
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:712:7: note: candidates are:
In file included from /usr/local/include/ceres/internal/autodiff.h:145:0,
                 from /usr/local/include/ceres/autodiff_cost_function.h:132,
                 from /usr/local/include/ceres/ceres.h:37,
                 from /home/either/catkin_tws/src/A-LOAM/src/lidarFactor.hpp:4,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:57:
/usr/local/include/ceres/jet.h:416:1: note: template<class T, int N> bool ceres::operator<(const ceres::Jet<T, N>&, const T&)
 CERES_DEFINE_JET_COMPARISON_OPERATOR( <  )  // NOLINT
 ^
/usr/local/include/ceres/jet.h:416:1: note:   template argument deduction/substitution failed:
In file included from /usr/include/eigen3/Eigen/Geometry:40:0,
                 from /usr/include/pcl-1.7/pcl/point_cloud.h:47,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:41:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:712:7: note:   deduced conflicting types for parameter ‘const T’ (‘double’ and ‘int’)
   if(d<0) scale1 = -scale1;
       ^
In file included from /usr/local/include/ceres/internal/autodiff.h:145:0,
                 from /usr/local/include/ceres/autodiff_cost_function.h:132,
                 from /usr/local/include/ceres/ceres.h:37,
                 from /home/either/catkin_tws/src/A-LOAM/src/lidarFactor.hpp:4,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:57:
/usr/local/include/ceres/jet.h:416:1: note: template<class T, int N> bool ceres::operator<(const T&, const ceres::Jet<T, N>&)
 CERES_DEFINE_JET_COMPARISON_OPERATOR( <  )  // NOLINT
 ^
/usr/local/include/ceres/jet.h:416:1: note:   template argument deduction/substitution failed:
In file included from /usr/include/eigen3/Eigen/Geometry:40:0,
                 from /usr/include/pcl-1.7/pcl/point_cloud.h:47,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:41:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:712:7: note:   mismatched types ‘const ceres::Jet<T, N>’ and ‘int’
   if(d<0) scale1 = -scale1;
       ^
In file included from /usr/local/include/ceres/internal/autodiff.h:145:0,
                 from /usr/local/include/ceres/autodiff_cost_function.h:132,
                 from /usr/local/include/ceres/ceres.h:37,
                 from /home/either/catkin_tws/src/A-LOAM/src/lidarFactor.hpp:4,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:57:
/usr/local/include/ceres/jet.h:416:1: note: template<class T, int N> bool ceres::operator<(const ceres::Jet<T, N>&, const ceres::Jet<T, N>&)
 CERES_DEFINE_JET_COMPARISON_OPERATOR( <  )  // NOLINT
 ^
/usr/local/include/ceres/jet.h:416:1: note:   template argument deduction/substitution failed:
In file included from /usr/include/eigen3/Eigen/Geometry:40:0,
                 from /usr/include/pcl-1.7/pcl/point_cloud.h:47,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserOdometry.cpp:41:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:712:7: note:   mismatched types ‘const ceres::Jet<T, N>’ and ‘int’
   if(d<0) scale1 = -scale1;
       ^
make[2]: *** [A-LOAM/CMakeFiles/alaserOdometry.dir/src/laserOdometry.cpp.o] 错误 1
make[1]: *** [A-LOAM/CMakeFiles/alaserOdometry.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
In file included from /usr/include/eigen3/Eigen/Geometry:40:0,
                 from /usr/include/pcl-1.7/pcl/point_cloud.h:47,
                 from /usr/include/pcl-1.7/pcl/conversions.h:50,
                 from /opt/ros/indigo/include/pcl_conversions/pcl_conversions.h:44,
[... 42 lines cut ...]
       ^
In file included from /usr/local/include/ceres/internal/autodiff.h:145:0,
                 from /usr/local/include/ceres/autodiff_cost_function.h:132,
                 from /usr/local/include/ceres/ceres.h:37,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserMapping.cpp:54:
/usr/local/include/ceres/jet.h:416:1: note: template<class T, int N> bool ceres::operator<(const ceres::Jet<T, N>&, const ceres::Jet<T, N>&)
 CERES_DEFINE_JET_COMPARISON_OPERATOR( <  )  // NOLINT
 ^
/usr/local/include/ceres/jet.h:416:1: note:   template argument deduction/substitution failed:
In file included from /usr/include/eigen3/Eigen/Geometry:40:0,
                 from /usr/include/pcl-1.7/pcl/point_cloud.h:47,
                 from /usr/include/pcl-1.7/pcl/conversions.h:50,
                 from /opt/ros/indigo/include/pcl_conversions/pcl_conversions.h:44,
                 from /home/either/catkin_tws/src/A-LOAM/src/laserMapping.cpp:43:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:712:7: note:   mismatched types ‘const ceres::Jet<T, N>’ and ‘int’
   if(d<0) scale1 = -scale1;
       ^
make[2]: *** [A-LOAM/CMakeFiles/alaserMapping.dir/src/laserMapping.cpp.o] 错误 1
make[1]: *** [A-LOAM/CMakeFiles/alaserMapping.dir/all] 错误 2
make: *** [all] 错误 2
Invoking "make -j4 -l4" failed
either@ubuntu:~/catkin_tws$ 
```

thank you!

getupgetup, 2019-06-13:
I've got the same error in Ubuntu14.04.
Here's my way to solve it:
sudo vi  /usr/include/eigen3/Eigen/src/Geometry/Quaternion.h
Change "if(d<0)" in line 712 to "if(d<Scalar(0))" and save.
Then you can use catkin_make to compile the project.
Hope it's useful to you.

wykxwyc, 2019-06-14:
> I've got the same error in Ubuntu14.04.
> Here's my way to solve it:
> sudo vi /usr/include/eigen3/Eigen/src/Geometry/Quaternion.h
> Change "if(d<0)" in line 712 to "if(d<Scalar(0))" and save.
> Then you can use catkin_make to compile the project.
> Hope it's useful to you.

Thank you very much!

closed 2019-06-14 by wykxwyc