HKUST-Aerial-Robotics/A-LOAM #7: Meaning of LidarEdgeFactor

getupgetup, 2019-06-13:
Hi, I'm confused about the residual computed by LidarEdgeFactor in lidarFactor.hpp:
Eigen::Matrix<T, 3, 1> nu = (lp - lpa).cross(lp - lpb);
Eigen::Matrix<T, 3, 1> de = lpa - lpb;
residual[0] = nu.x() / de.norm();
residual[1] = nu.y() / de.norm();
residual[2] = nu.z() / de.norm();
What's the meaning of this residual? 
Since the residual of LidarPlaneFactor means the distance from the point to the plane, shouldn't  the residual of LidarEdgeFactor mean the distance from the point to the line? How to explain that  
the residual is 3-dimensional? Thank you~

andre-nguyen, 2019-06-13:
Isn't it this equation? https://en.wikipedia.org/wiki/Distance_from_a_point_to_a_line#Another_vector_formulation

getupgetup, 2019-06-14:
> Isn't it this equation? https://en.wikipedia.org/wiki/Distance_from_a_point_to_a_line#Another_vector_formulation

Yeah, I can find the same equation in Ji.Zhang's paper. However, the distance is a scalar instead of a 3-d vector. Although both are same to Ceres, I still can't understand the meaning of vector instead of distance.

lizimo061, 2019-06-16:
I think the residual here is most relevant to the Jacobian computation. Distance is not directly used in the cost function if you write the Jacobian equations explicitly for the point to LiDAR edge distance.

getupgetup, 2019-06-17:
> I think the residual here is most relevant to the Jacobian computation. Distance is not directly used in the cost function if you write the Jacobian equations explicitly for the point to LiDAR edge distance.

However, the residual is just the distance in LidarPlaneFactor.

closed 2022-01-10 by getupgetup