HKUST-Aerial-Robotics/A-LOAM #8: terminate called after throwing an instance of 'std::bad_alloc'

LongruiDong, 2019-06-28:
I have run successfully to get result  of KITTI 00~09. But when I test Sequence 10, Error happens:
```
ThinkStation:~/catkin_ws$ roslaunch aloam_velodyne kitti_helper.launch
... logging to /home/*/.ros/log/22327ae4-994a-11e9-b2df-cc52af3d9390/roslaunch-dlr-ThinkStation-14860.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ThinkStation:34326/

SUMMARY
========

PARAMETERS
 * /kittiHelper/dataset_folder: /home/*/Project...
 * /kittiHelper/output_bag_file: /home/*/catkin_...
 * /kittiHelper/publish_delay: 1
 * /kittiHelper/sequence_number: 10
 * /kittiHelper/to_bag: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    kittiHelper (aloam_velodyne/kittiHelper)

ROS_MASTER_URI=http://localhost:11311

process[kittiHelper-1]: started with pid [14877]
Reading sequence 10 from /home/*/Project/ORB_SLAM2/Examples/kitti-benchmark/dataset/
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[kittiHelper-1] process has died [pid 14877, exit code -6, cmd /home/*/catkin_ws/devel/lib/aloam_velodyne/kittiHelper __name:=kittiHelper __log:=/home/*/.ros/log/22327ae4-994a-11e9-b2df-cc52af3d9390/kittiHelper-1.log].
log file: /home/*/.ros/log/22327ae4-994a-11e9-b2df-cc52af3d9390/kittiHelper-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Could you help me with it?
Thanks in advance~

LongruiDong, 2019-06-28:
Oops, I know it, I haven't Sequence 10 data, 
Sorry to bother 23333

closed 2019-06-28 by LongruiDong

xuwuzhou, 2020-01-10:
@LongruiDong HI,
I met a similar problem,last week,I generate the bag successfully,but today,it says"
Reading sequence 04 from /home/xuwuzhou/dataset/
totally 124231 points in this lidar frame 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
"
Obviously,"totally 124231 points in this lidar frame "means the data is there,so what happened?

GarethZhang, 2021-02-23:
I changed `velodyne/sequences/` on Line 132 in kittiHelper.cpp to `sequences/` to make this work