HKUST-Aerial-Robotics/A-LOAM #10: compile error

mutou-2015, 2019-07-16:
hi,when I compile A-LOAM package,I meet some errors as fellow,

/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Solve(cs_di_symbolic*, cs_di_numeric*, double*)':
cxsparse.cc:(.text+0x92): undefined reference to `cs_di_ipvec'
cxsparse.cc:(.text+0xa6): undefined reference to `cs_di_lsolve'
cxsparse.cc:(.text+0xba): undefined reference to `cs_di_ltsolve'
cxsparse.cc:(.text+0xd3): undefined reference to `cs_di_pvec'
cxsparse.cc:(.text+0x108): undefined reference to `cs_di_free'
cxsparse.cc:(.text+0x114): undefined reference to `cs_di_malloc'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::SolveCholesky(cs_di_sparse*, double*)':
cxsparse.cc:(.text+0x31a): undefined reference to `cs_di_cholsol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::BlockAnalyzeCholesky(cs_di_sparse*, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&)':
cxsparse.cc:(.text+0x438): undefined reference to `cs_di_amd'
cxsparse.cc:(.text+0x4ba): undefined reference to `cs_di_free'
cxsparse.cc:(.text+0x4f6): undefined reference to `cs_di_calloc'
cxsparse.cc:(.text+0x506): undefined reference to `cs_di_pinv'
cxsparse.cc:(.text+0x516): undefined reference to `cs_di_symperm'
cxsparse.cc:(.text+0x523): undefined reference to `cs_di_etree'
cxsparse.cc:(.text+0x532): undefined reference to `cs_di_post'
cxsparse.cc:(.text+0x546): undefined reference to `cs_di_counts'
cxsparse.cc:(.text+0x551): undefined reference to `cs_di_free'
cxsparse.cc:(.text+0x559): undefined reference to `cs_di_spfree'
cxsparse.cc:(.text+0x569): undefined reference to `cs_di_malloc'
cxsparse.cc:(.text+0x57b): undefined reference to `cs_di_cumsum'
cxsparse.cc:(.text+0x58d): undefined reference to `cs_di_free'
cxsparse.cc:(.text+0x60c): undefined reference to `cs_di_sfree'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::CreateSparseMatrix(ceres::internal::TripletSparseMatrix*)':
cxsparse.cc:(.text+0x802): undefined reference to `cs_di_compress'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::ApproximateMinimumDegreeOrdering(cs_di_sparse*, int*)':
cxsparse.cc:(.text+0x890): undefined reference to `cs_di_amd'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::~CXSparse()':
cxsparse.cc:(.text+0x2b): undefined reference to `cs_di_free'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Cholesky(cs_di_sparse*, cs_di_symbolic*)':
cxsparse.cc:(.text+0x47): undefined reference to `cs_di_chol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::AnalyzeCholesky(cs_di_sparse*)':
cxsparse.cc:(.text+0x336): undefined reference to `cs_di_schol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::AnalyzeCholeskyWithNaturalOrdering(cs_di_sparse*)':
cxsparse.cc:(.text+0x343): undefined reference to `cs_di_schol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::ApproximateMinimumDegreeOrdering(cs_di_sparse*, int*)':
cxsparse.cc:(.text+0x8b7): undefined reference to `cs_di_free'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::TransposeMatrix(cs_di_sparse*)':
cxsparse.cc:(.text+0x8c9): undefined reference to `cs_di_transpose'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::MatrixMatrixMultiply(cs_di_sparse*, cs_di_sparse*)':
cxsparse.cc:(.text+0x8d7): undefined reference to `cs_di_multiply'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Free(cs_di_sparse*)':
cxsparse.cc:(.text+0x8e4): undefined reference to `cs_di_spfree'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Free(cs_di_symbolic*)':
cxsparse.cc:(.text+0x8f4): undefined reference to `cs_di_sfree'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Free(cs_di_numeric*)':
cxsparse.cc:(.text+0x904): undefined reference to `cs_di_nfree'
collect2: error: ld returned 1 exit status
A-LOAM/CMakeFiles/alaserOdometry.dir/build.make:437: recipe for target '/home/nari/catkin_ws/devel/lib/aloam_velodyne/alaserOdometry' failed
make[2]: *** [/home/nari/catkin_ws/devel/lib/aloam_velodyne/alaserOdometry] Error 1
CMakeFiles/Makefile2:2458: recipe for target 'A-LOAM/CMakeFiles/alaserOdometry.dir/all' failed
make[1]: *** [A-LOAM/CMakeFiles/alaserOdometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Solve(cs_di_symbolic*, cs_di_numeric*, double*)':
cxsparse.cc:(.text+0x92): undefined reference to `cs_di_ipvec'
cxsparse.cc:(.text+0xa6): undefined reference to `cs_di_lsolve'
cxsparse.cc:(.text+0xba): undefined reference to `cs_di_ltsolve'
cxsparse.cc:(.text+0xd3): undefined reference to `cs_di_pvec'
cxsparse.cc:(.text+0x108): undefined reference to `cs_di_free'
cxsparse.cc:(.text+0x114): undefined reference to `cs_di_malloc'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::SolveCholesky(cs_di_sparse*, double*)':
cxsparse.cc:(.text+0x31a): undefined reference to `cs_di_cholsol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::BlockAnalyzeCholesky(cs_di_sparse*, std::vector<int, std::allocator<int> > const&, std::vector<int, std::allocator<int> > const&)':
cxsparse.cc:(.text+0x438): undefined reference to `cs_di_amd'
cxsparse.cc:(.text+0x4ba): undefined reference to `cs_di_free'
cxsparse.cc:(.text+0x4f6): undefined reference to `cs_di_calloc'
cxsparse.cc:(.text+0x506): undefined reference to `cs_di_pinv'
[... 18 lines cut ...]
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::AnalyzeCholesky(cs_di_sparse*)':
cxsparse.cc:(.text+0x336): undefined reference to `cs_di_schol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::AnalyzeCholeskyWithNaturalOrdering(cs_di_sparse*)':
cxsparse.cc:(.text+0x343): undefined reference to `cs_di_schol'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::ApproximateMinimumDegreeOrdering(cs_di_sparse*, int*)':
cxsparse.cc:(.text+0x8b7): undefined reference to `cs_di_free'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::TransposeMatrix(cs_di_sparse*)':
cxsparse.cc:(.text+0x8c9): undefined reference to `cs_di_transpose'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::MatrixMatrixMultiply(cs_di_sparse*, cs_di_sparse*)':
cxsparse.cc:(.text+0x8d7): undefined reference to `cs_di_multiply'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Free(cs_di_sparse*)':
cxsparse.cc:(.text+0x8e4): undefined reference to `cs_di_spfree'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Free(cs_di_symbolic*)':
cxsparse.cc:(.text+0x8f4): undefined reference to `cs_di_sfree'
/usr/local/lib/libceres.a(cxsparse.cc.o): In function `ceres::internal::CXSparse::Free(cs_di_numeric*)':
cxsparse.cc:(.text+0x904): undefined reference to `cs_di_nfree'
collect2: error: ld returned 1 exit status
A-LOAM/CMakeFiles/alaserMapping.dir/build.make:437: recipe for target '/home/nari/catkin_ws/devel/lib/aloam_velodyne/alaserMapping' failed
make[2]: *** [/home/nari/catkin_ws/devel/lib/aloam_velodyne/alaserMapping] Error 1
CMakeFiles/Makefile2:1963: recipe for target 'A-LOAM/CMakeFiles/alaserMapping.dir/all' failed
make[1]: *** [A-LOAM/CMakeFiles/alaserMapping.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

Can you give me some advices how to resolve these problems?Thank you!

shinmohuang, 2022-05-12:
I've got the same problem with you. Have you sloved? thx

Kin-Zhang, 2022-11-11:
maybe using the ceres 1.14 will help that.

```bash
cd / && git clone --depth 1 --branch 1.14.0 https://ceres-solver.googlesource.com/ceres-solver \
    && cd ceres-solver && mkdir build && cd build && cmake .. \
    && make -j$(nproc) \
    && make -j$(nproc) all install \
    && cd / \
    && rm -rf /ceres-solver
```