HKUST-Aerial-Robotics/A-LOAM #11: Question about kittiHelper.launch

Gatsby23, 2019-07-22:
I have download the Kitti Velodyne bag files into my data folder, but it doesn't work. Should I download the color sequences or the gray sequency to the same folder?

zhu-bingo, 2019-11-06:
hello,i meet the same problem about kittihelper.launch,have you solved it? Thank you

Juuustin, 2020-04-26:
same question

miroslavradojevic, 2020-07-04:
Hello, the same question here. According to the manual, if one wants to try KITTI demo, it is necessary to: 
- download KITTI Odometry dataset to YOUR_DATASET_FOLDER 

It is not stated whether all datasets gray/color/lidar are needed or not, and what exactly corresponds to YOUR_DATASET_FOLDER? Which directory of the folder structure is needed here? There are data_odometry_X.zip files (X=gray,color,velodyne) with  dataset/sequences/00 underneath 
Which folder is used as YOUR_DATASET_FOLDER in kitti_helper.launch file?

It is not explained. I tried adding:
$UNZIP_LOC/
$UNZIP_LOC/dataset/
$UNZIP_LOC/dataset/sequences/

and providing with sequence number 00, 01, etc.. 
It does not seem to be reading at all. Could you provide some hints on this?

closed 2020-07-09 by Gatsby23