HKUST-Aerial-Robotics/A-LOAM #13: why not transform feature points to the scan end point

zorosmith, 2019-08-12:
Hello, thank you for your great work!
I found that you disenabled the operation to transform corner features and plane features to the scan end point.
I enabled this operation and run the project again. However, I found nothing changed. Maybe I missed something. 
Could you tell me why would you disenable this operation?

wwtinwhu, 2019-08-14:
I suppose it depends on the scan has distortion or not. If has no distortion, like kitti data, disenable.

mykolas3, 2019-10-08:
> I suppose it depends on the scan has distortion or not. If has no distortion, like kitti data, disenable.

hi, i was wondering what distortion is removed from kitti? please correct me if i'm wrong but I guess the setup is some velodyne lidar (say, with 16 lasers) that are on a same vertical line and it's constantly being rotated around the up axis to cover 360degree horizontal FOV. the 16 lasers do not all precisely pulse at the same time and also the lidar generates 16 horizontal scanlines per frame. so does kitti remove the distortion per horizontal scanlines so you get 16 separate undistorted scanlines or vertically - as if all 16 lasers pulse at the same time? or is it something completely different? thanks.

LarryDong, 2024-11-17:
I recently checked this code, and find that undistortion needs two modification. 

The first is, to set `DISTORTION` to 1, in `laserOdometry.cpp`, without this, the undistortion scalar `s` is always 1 so no undistrotion.
```cpp
#define DISTORTION 0
```
However, this only changed the kdtree search process, but doesn't necessarily change the original pointcloud. 

The second is, at the end of this file, there is:
```cpp
// transform corner features and plane features to the scan end point
if (0)
{
    int cornerPointsLessSharpNum = cornerPointsLessSharp->points.size();
    for (int i = 0; i < cornerPointsLessSharpNum; i++)
    {
        TransformToEnd(&cornerPointsLessSharp->points[i], &cornerPointsLessSharp->points[i]);
    }
...
```
the if(0) should be changed to "if(1)", this undistort all points to the end of the scan (however, that is also based on TransformToStart). Without this enabled, the corner/flat points are not changed to the next node.

If didnot know why the code is like this, but I changed them, and the undistortion seems correct (under constant motion cases).

This issue has been a long time, but I hope this reply helps other.

closed 2024-12-01 by zorosmith