HKUST-Aerial-Robotics/A-LOAM #14: Question about Kitti data.

Gatsby23, 2019-08-27:
The results from rviz I can get is this:
[image]
and I have already select after_maped topic. Is there anything wrong?
I also confused that I can't find my point cloud image and i have select the point cloud topic too? I realy want to know how to recurring the effect like this ?
[image]
I really need your help, Is there anything wrong, could you please give me some advice ? Thanks!

LongruiDong, 2019-09-16:
> The results from rviz I can get is this:
> [image]
> and I have already select after_maped topic. Is there anything wrong?
> I also confused that I can't find my point cloud image and i have select the [...]

you can select "/image_left" in the **image** _topic_ , then image window will show image sequences;

However I  am also confusing about how to display that _moving red car_ in above example picture, anyone knows?

Gatsby23, 2019-09-16:
> you can select "/image_left" in the **image** _topic_ , then image window will show image sequences;
>
> However I am also confusing about how to display that _moving red car_ in above example picture, [...]

Thank you for your replay. I have already know how to view the image sequences. The problem is that I thought the odometry path is too low to believe. Another thing is why the colorful points doesn't show anymore ?

zhu-bingo, 2019-11-06:
can you tell me how to run a-loam with kitti odometry dataset

dhananjaisharma10, 2019-11-26:
> The results from rviz I can get is this:
> [image]
> and I have already select after_maped topic. Is there anything wrong?
> I also confused that I can't find my point cloud image and i have select the [...]

Hi! Could you please tell me how to run A-LOAM on the KITTI odometry dataset in a bit detail. I followed the README instructions but couldn't get it to run. Please let me know. Thanks!

mhaboali, 2020-04-16:
For the future debuggers, 

Try to debug a little bit the values of the paths inside kittihelper.cpp node. After fixing those paths you'll be able to run the example with KITTI odometry data exactly as it's shown in readme file.

Muhammad,
Thanks

Gatsby23, 2020-04-16:
> For the future debuggers,
> 
> Try to debug a little bit the values of the paths inside kittihelper.cpp node. After fixing those paths you'll be able to run the example with KITTI odometry data exactly as it's shown in readme file.
> 
> Muhammad,
> Thanks

OK, Thank you a lot

closed 2020-04-16 by Gatsby23

Juuustin, 2020-04-26:
> For the future debuggers,
> 
> Try to debug a little bit the values of the paths inside kittihelper.cpp node. After fixing those paths you'll be able to run the example with KITTI odometry data exactly as it's shown in readme file.
> 
> Muhammad,
> Thanks

Could you write more for debugging?

mhaboali, 2020-05-01:
Open the kittiplayer.cpp file and check the paths for poses, images, and lidar folders. Let me know if it's still unclear.

Thanks,
Mohammed