HKUST-Aerial-Robotics/A-LOAM #18: Unstable Of Distortion In LaserOdometry

johnsongash, 2019-10-15:
Thanks for the nice code from your guys,
after studying the detail of it there is one question here.

In LaserOdometry since Kitti provide undistortion pointcloud, distortion is turn off in the code.
But for our own data(VLP16), turning on distortion to true makes the result much poor, the path oscillate and sometimes diverge, we have notice that TransformToEnd shoud turn on too...
Pictures of turning on distortion(Upper) and turning off(Lower) for same data are shown below, any advice here?

[image]

[image]

nikwitting, 2019-11-09:
I just want to confirm that I experience the same problem with the gates_oscillating_motion.bag dataset with distortion=1 and TransformToEnd activated. Did you find a solution?
[image]

johnsongash, 2019-11-10:
@nikwitting yes the problem still exists, not finding solution yet.
@shaozu hope the author help us out.

ZhenminHuang, 2020-01-06:
@johnsongash @nikwitting Not sure if it's too late to make this comment. I also met the same problem. I tried replace the loss function HuberLoss(0.1) to CauchyLoss(0.05) and that killed down the oscillation by considerable amount (although not completely). Perhaps you can try other combinations of loss functions and coefficients?

This is the path if HuberLoss(0.1) was used.
[image]

This is the path if CauchyLoss(0.05) was used.
[image]

Hope I help.

manymuch, 2021-01-25:
I encountered the same problem as mentioned by @johnsongash @nikwitting, and @ZhenminHuang 's solution does mitigate the oscillating problem but still can not fully solved the issue.
Another walk-around is by comment out [L28](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/devel/src/lidarFactor.hpp#L28) and [L29](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/devel/src/lidarFactor.hpp#L29) and of course in L80 and L81.
I guess there might be some problem with interpolation of quaternion using ceres auto differentiation.
Note that the author might have noticed similar issue, and leave [L26](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/devel/src/lidarFactor.hpp#L26), which is a approximate of quaternion slerp when rotation is small enough, which is, quite a common assumption for ground driving vehicles.

h-k8888, 2021-11-22:
in the file laserOdometry.cpp, turning on distortion in the function [TransformToStar](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/e51f88c12b483ce42dfa219783712c7234dc8c08/src/laserOdometry.cpp#L115) and the function [transformToEnd](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/e51f88c12b483ce42dfa219783712c7234dc8c08/src/laserOdometry.cpp#L133) is the way to call distortion correction. 
There is no need to correct the distortion twice by turning on DISTORTION in [line](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/e51f88c12b483ce42dfa219783712c7234dc8c08/src/laserOdometry.cpp#L376) and [plane](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/e51f88c12b483ce42dfa219783712c7234dc8c08/src/laserOdometry.cpp#L474) when doing registration.
And the output is better.
Is that right?

bryantaoli, 2022-10-21:
I had the same problem. I think it is because the assumption of uniform velocity is not valid, leading to the failure of deskew. I solved this problem by taking the distorted value and using the result after the first scan2scan to deskew.