HKUST-Aerial-Robotics/A-LOAM #20: KITTI dataset to bag file

zhu-bingo, 2019-10-31:
how to convert KITTI dataset to bag file ?
thank you

dhananjaisharma10, 2019-11-26:
> how to convert KITTI dataset to bag file ?
> thank you

Hi! Have you resolved your issue? Please let me know. Thanks.

xuwuzhou, 2020-01-04:
kitti2bag can do it

Juuustin, 2020-04-26:
> kitti2bag can do it

Could you give me a link? I tried several times, there are always errors.

Juuustin, 2020-04-28:
> > how to convert KITTI dataset to bag file ?
> > thank you
> 
> Hi! Have you resolved your issue? Please let me know. Thanks.

Did you solve it?

kissb2, 2020-06-20:
without making a bulky bag file, I wrote a simple python script to publish the kitti files. https://github.com/kissb2/mini-kitti-publisher
I hope it could help you.

MrNeoJeep, 2021-07-31:
@zhu-bingo you can use kitti_to_rosbag or kitti2bag