HKUST-Aerial-Robotics/A-LOAM #21: Saving the mapping result

bennywijaya55, 2020-01-14:
Hi, first of all I want to say great work on this algorithm, very robust and useful!
My question here is after we run the algorithm, and we have done the mapping process, how to save this map information into a file?

Thanks  a lot!

aschenck, 2020-02-18:
I save the data of the topic /laser_cloud_map to a bagfile and read it out using matlab. I extract the last message using the 'readXYZ' function and convert that to a 'pointCloud' object. I can then display it using the 'pcshow' plot function.

Juuustin, 2020-05-09:
> I save the data of the topic /laser_cloud_map to a bagfile and read it out using matlab. I extract the last message using the 'readXYZ' function and convert that to a 'pointCloud' object. I can then display it using the 'pcshow' plot function.

Hi, I also save the data of the topic /laser_cloud_map, but it looks like a input of each frame, could you explain it more? thank you very much!

aschenck, 2020-05-10:
afaik it's the same data you see when using rviz while mapping live. Every time the map gets updated it gets published to /laser_cloud_map.  So when you save the data in a rosbag, it's only the last message that contains the entire map.

Juuustin, 2020-05-10:
> afaik it's the same data you see when using rviz while mapping live. Every time the map gets updated it gets published to /laser_cloud_map. So when you save the data in a rosbag, it's only the last message that contains the entire map.
> There are also some settings in the launch file that increases the level of detail of the published maps.
> param name="mapping_line_resolution" type="double" value="0.4"/
> param name="mapping_plane_resolution" type="double" value="0.8"/
> 
> lowering these values should provide nicer maps. gl

Make sense, and thank you very much!

bennywijaya55, 2020-10-22:
Thank you for your reply, really helpful! 
Btw, I forgot to close this thread!

closed 2020-10-22 by bennywijaya55