HKUST-Aerial-Robotics/A-LOAM #22: How to use ouster 64 line lidar?

LeisureLei, 2020-02-03:
I want to run A-LOAM with ouster OS1 64 ,but I am confused with mapping_line_resolution and mapping_plane_resolution. How can I set these parameters?

aschenck, 2020-02-18:
I'm using an os1-64 as well, all I did was remap the os1 laserscan topic to /velodyne_points and used the velodyne64 launch file included in a-loam. Results seem good thus far.

kissb2, 2020-06-19:
@LeisureLei 
This would be help.
https://youtu.be/Ea2sDqXf6ZY

JzHuai0108, 2023-11-16:
The scanID is simply the ring number which is always provided by modern lidars, e.g., VLP16, Hesai Pandar, Ouster OS1 from our experience.

mar-watt, 2024-11-23:
> > 我也在使用 os1-64，我所做的就是将 os1 laserscan 主题重新映射到 /velodyne_points，并使用 a-loam 中包含的 velodyne64 启动文件。到目前为止，结果似乎不错。
> 
> I used the remap method to run Ouster64 data, but in RViz, the point cloud frame and the map are vertical. Do you know what the problem is?

Haven't had that issue myself, maybe a static transform adding rotation to the points?