HKUST-Aerial-Robotics/A-LOAM #28: Problem on kitti_helper.launch

Juuustin, 2020-04-28:
Now I am trying to test A-LOAM on kitti (velodyne laser data, 80 GB), as far as I know, we should try to transform them into ROS bag at first. (if it is not true, please point it out).

When I use the `kitti_helper.launch`, the output is only a .bag file about 4kB in total.

This is the terminal information:
`... logging to /home/objloc/.ros/log/c787383e-8951-11ea-b5c0-9cebe8b95aff/roslaunch-deeplearning-12255.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://deeplearning:42849/

SUMMARY
========

PARAMETERS
 * /kittiHelper/dataset_folder: /home/objloc/OBJL...
 * /kittiHelper/output_bag_file: /home/objloc/OBJL...
 * /kittiHelper/publish_delay: 1
 * /kittiHelper/sequence_number: 004
 * /kittiHelper/to_bag: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    kittiHelper (aloam_velodyne/kittiHelper)

auto-starting new master
process[master]: started with pid [12267]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c787383e-8951-11ea-b5c0-9cebe8b95aff
process[rosout-1]: started with pid [12282]
started core service [/rosout]
process[kittiHelper-2]: started with pid [12300]
Reading sequence 004 from /home/objloc/OBJLOC/A-LOAM/
Done 
[kittiHelper-2] process has finished cleanly
log file: /home/objloc/.ros/log/c787383e-8951-11ea-b5c0-9cebe8b95aff/kittiHelper-2*.log
`
any suggestion or advice for it?

songming1995, 2020-10-10:
Hi, I met the same problem. Then I check the kittiHelp.cpp and set the correct path of my dataset in the launch file. Pay attention, the path should end with a '/' to show it is a folder not a file. Example:
 <param name="dataset_folder" type="string" value="/home/songming/Desktop/dataset/" /> 
Another way to check if it is your path is correct is to cout the path string in the source code of 'kittiHelp.cpp' in case of ant misspelling. 
Hope it will help you.

RG2806, 2021-10-05:
@Juuustin were you able solve this.
I have encountered the same problem. @songming1995's solution didn't help

yangfuwei, 2022-03-10:
Hi, I met the same problem and I found that "lidar_data_path" in KittiHelp.cpp no longer match the path in KITTI dataset.