HKUST-Aerial-Robotics/A-LOAM #31: Quanergy M8

danbo2, 2020-05-03:
Hi, if I want to use the data collected from Quanergy M8 Lidar, what should I do?  I find the map can not be generated when I play the data collected with Quanergy M8. Thanks

cristianrubioa, 2020-10-21:
Hi @danbo2 have you found a way to do it with the Quanergy M8 Lidar?

danbo2, 2020-10-21:
Yes, you can use node transition method in ROS and I encourage you to learn more about ROS. I just add one line 'rosrun topic_tools throttle messages /Sensor/points 10 /velodyne_points'. It means I convert /Sensor/point 10 to /velodyne points.

cristianrubioa, 2020-10-21:
Thanks @danbo2 I can already generate the map. I can ask you what additional parameters you set for this Lidar in the algorithm?

danbo2, 2020-10-22:
I don't think I changed any parameters.

cristianrubioa, 2020-10-22:
I mean adapt-> https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/devel/src/scanRegistration.cpp for this specific QM8 Lidar

danbo2, 2020-10-22:
No, I didn't make any change of the original source code.