HKUST-Aerial-Robotics/A-LOAM #35: Huge vertical deviation for a long dataset

mhaboali, 2020-05-27:
Hi thanks for sharing such a good project. 

I've tested your project with my own dataset which is around 1700 seconds. What I noticed there was a huge vertical deviation for the path after around 1300 seconds as shown below

**allMapCloud**
[image]

**AftMapped Path**
It's very clear here, that's supposed to be at the same level.
[image]

And the ground truth doesn't have this vertical deviation at all, all streets were at the same level.

What do you think the issue would come from? And how can I resolve this issue?

Thanks!

mhaboali, 2020-05-27:
Hi @qintonguav,
Please let me know what do you think of this issue?

Thanks!

kissb2, 2020-06-20:
@mhaboali 
Actually, for the large-scale env, the phenomenon is inevitable I think. You can consider loop detection and loop closing of a pose-graph (e.g., https://github.com/irapkaist/SC-LeGO-LOAM could be a solution)

mhaboali, 2020-06-23:
Thanks @kissb2 for your recommendation, I'll try it shortly and I'll keep you posted on the status.

Thanks again and I appreciate your help!
Mohammad

EdCheriee, 2020-06-29:
I have a similar issue on larger datasets (beyond 1200s) and have the same issue that and doesn't detect the height levels right (when returning to the same point it doesn't recognize the same ground level. Let me know if you had any success (trying it myself as well).

mhaboali, 2020-06-29:
Yeah @EdCheriee , my dataset is around 1700 seconds. Actually, I didn't manage to resolve this issue with A-Loam but I'm still searching for another alternative. Let's keep the issue here updated with our findings and let's hope we'll find a proper alternative.

papicheng, 2021-01-05:
帮顶，同遇到这个问题，难道原作者是忽略这个问题了吗[笑脸]