HKUST-Aerial-Robotics/A-LOAM #39: Vertical velodyne sensor setup for a drone simulation

didzdissanayaka8, 2020-11-10:
Hi, I want to mount the Velodyne VLP 16 sensor vertically for my simulations as shown below. However, when I tried the A-LOAM package with that sensor setup, the state estimation and mapping were not accurate enough. I did not change any coordinate frames in this case. Does anyone know whether this is possible and whether anything needs to be changed in the code? 
[image]

Gyudori, 2021-01-07:
@didzdissanayaka8 
I think the ratio of overlap for consecutive frames might be cause that result. Lidar data frames horizontally installed on a vehicle overlap for many areas, but in your setup on a drone does not.

Ahmad-Drak, 2022-10-04:
@didzdissanayaka8 did you find a feasible solution for this? currently facing the same issue.

pie-leroy, 2024-07-17:
any update @didzdissanayaka8  ? i am looking for a solution

Ahmad-Drak, 2024-07-17:
@pie-leroy sadly nothing yet. I spent too much time on this and I needed to move on..Would still be interested in a solution really.