HKUST-Aerial-Robotics/A-LOAM #44: scanRegistration issue

huhu965, 2021-08-02:
if (!halfPassed)
        {   //确保-pi/2 < ori - startOri < 3*pi/2
            if (ori < startOri - M_PI / 2)
            {
                ori += 2 * M_PI;
            }
            else if (ori > startOri + M_PI * 3 / 2)
            {
                ori -= 2 * M_PI;
            }

            if (ori - startOri > M_PI)
            {
                halfPassed = true;
            }
        }
        else
        {   //确保-3*pi/2 < ori - endOri < pi/2
            ori += 2 * M_PI;
            if (ori < endOri - M_PI * 3 / 2)
            {
                ori += 2 * M_PI;
            }
            else if (ori > endOri + M_PI / 2)
            {
                ori -= 2 * M_PI;
            }
        }

        float relTime = (ori - startOri) / (endOri - startOri); 

您好，我想问一下关于这块的公式是正确的吗？M_PI < endOri - startOri < 3 * M_PI，-pi/2 < ori - startOri < 3*pi/2
我发现计算出的结果是错误的，我不知道是不是我的理解有问题，能帮吗解答一下吗？在我理解中，应该将ori按照endori的方式，加2π，然后再分类讨论，才能得出正确的时间结果。

Irrman, 2021-09-13:
我也有相同的 疑问

Liao-gl, 2023-06-22:
我也有疑问，请问理解了吗