HKUST-Aerial-Robotics/A-LOAM #45: No point cloud, only trajectory result output in rviz.

ZhangZh3ng, 2021-11-10:
I got the following result  on my pc:
[image]
It seems the author's demo contains more information, such as point cloud:
[image]

what should i do to display these information in rviz window?

closed 2021-11-17 by ZhangZh3ng

cybaol, 2022-01-15:
Hi bro, I also encountered the same problem, how did you solve it ?

chj0728, 2022-04-13:
> Hi bro, I also encountered the same problem, how did you solve it ?
 Delete the  "/" of "/ camera_init" in cpp file

chj0728, 2022-06-22:
> > > Hi bro, I also encountered the same problem, how did you solve it ?
> > > Delete the  "/" of "/ camera_init" in cpp file
> 
> Sry, which cpp file you mean? the kittihelper.cpp?

yep，also including laserMapping.cpp,laserOdometry.cpp,scanRegistration.cpp