HKUST-Aerial-Robotics/A-LOAM #47: Logic of the scanID

arjunskumar, 2022-02-17:
can you explain the logic of [L171](https://github.com/HKUST-Aerial-Robotics/A-LOAM/blob/e51f88c12b483ce42dfa219783712c7234dc8c08/src/scanRegistration.cpp#L171). 
I'm using VLP16 lidar, attaching my scanID and Vertcal angle relationship diagram 
[image]

smileyenot983, 2022-04-01:
I have printed angle to scanID correspondings and for me it is like that:

angle -14.999978 scanID 0 
angle -12.999980 scanID 1 
angle -10.999983 scanID 2 
angle -8.999986 scanID 3 
angle -6.999990 scanID 4 
angle -4.999993 scanID 5 
angle -2.999995 scanID 6 
angle -0.999998 scanID 7 
angle 0.999998 scanID 8 
angle 2.999996 scanID 9 
angle 4.999993 scanID 10 
angle 6.999990 scanID 11 
angle 8.999987 scanID 12 
angle 10.999983 scanID 13 
angle 12.999980 scanID 14 
angle 14.999978 scanID 15 

Looks like these scanIDs are only used for painting circles with different colors, when they assign intensity to point depending on it's scanID

arjunskumar, 2022-04-01:
@smileyenot983 which lidar you were using?

smileyenot983, 2022-04-01:
It is VLP-16, just like you

arjunskumar, 2022-04-01:
@smileyenot983 Thanks

closed 2022-09-26 by arjunskumar