HKUST-Aerial-Robotics/A-LOAM #54: Version Compatibility

Yigit-Kuyu, 2023-06-13:
Can this algorithm be implemented with Ubuntu 20.04, ROS neotic?

Yigit-Kuyu, 2023-06-22:
Thank you, I will try.

minglok943, 2023-07-25:
Hi, what is the version of pcl you are using?

minglok943, 2023-07-25:
I followed. But I met with this error. 
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libcudart_static.a', needed by '/home/####/catkin_ws/devel/lib/aloam_velodyne/alaserOdometry'.  Stop.
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libcudart_static.a', needed by '/home/####/catkin_ws/devel/lib/aloam_velodyne/alaserMapping'.  Stop.

my gpu is GTX-1660ti
I installed cuda 11.8 toolkit and nvidia 520 driver following [https://developer.nvidia.com/cuda-11-8-0-download-archive?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=20.04&target_type=deb_local](url)