HKUST-Aerial-Robotics/A-LOAM #65: catkin_make error

bason-ren, 2024-01-03:
/usr/bin/ld: cannot find -lCUDA::cublas
/usr/bin/ld: cannot find -lCUDA::cudart
/usr/bin/ld: cannot find -lCUDA::cusolver
/usr/bin/ld: cannot find -lCUDA::cusparse
collect2: error: ld returned 1 exit status

What should I do to solve this problem

SunZezhou, 2024-02-29:
In my personal experience, this is because the `USE_CUDA` parameter is turned on when compiling Ceres. Setting `USE_CUDA` to `OFF` in CMakeLists of Ceres and then re-install it may solve this problem.

mz7790, 2024-04-25:
The solution I encountered was similar to yours. I'm using Ceres 2.2.0. When compiling Ceres 2.2.0, everything goes smoothly. However, when compiling ALOAM, I encounter the same error as you. I found that copying the CUDA-related commands from the CMakeLists.txt file in the Ceres library resolved the issue.

Here's the content I copied:
···
option(USE_CUDA "Enable use of CUDA linear algebra solvers." ON)
if (USE_CUDA)
  if (CMAKE_VERSION VERSION_LESS 3.17)
    # On older versions of CMake (20.04 default is 3.16) FindCUDAToolkit was
    # not available, but FindCUDA was deprecated. To avoid special-case handling
    # elsewhere, emulate the effects of FindCUDAToolkit locally in terms of the
    # expected CMake imported targets and defined variables. This can be removed
    # from as soon as the min CMake version is >= 3.17.
    find_package(CUDA QUIET)
    if (CUDA_FOUND)
      message("-- Found CUDA version ${CUDA_VERSION} installed in: "
        "${CUDA_TOOLKIT_ROOT_DIR} via legacy (< 3.17) CMake module. "
        "Using the legacy CMake module means that any installation of "
        "Ceres will require that the CUDA libraries be installed in a "
        "location included in the LD_LIBRARY_PATH.")
      enable_language(CUDA)

      macro(DECLARE_IMPORTED_CUDA_TARGET COMPONENT)
        add_library(CUDA::${COMPONENT} INTERFACE IMPORTED)
        target_include_directories(
          CUDA::${COMPONENT} INTERFACE ${CUDA_INCLUDE_DIRS})
        target_link_libraries(
          CUDA::${COMPONENT} INTERFACE ${CUDA_${COMPONENT}_LIBRARY} ${ARGN})
      endmacro()

      declare_imported_cuda_target(cublas)
      declare_imported_cuda_target(cusolver)
      declare_imported_cuda_target(cusparse)
      declare_imported_cuda_target(cudart ${CUDA_LIBRARIES})

      set(CUDAToolkit_BIN_DIR ${CUDA_TOOLKIT_ROOT_DIR}/bin)

    else (CUDA_FOUND)
      message("-- Did not find CUDA, disabling CUDA support.")
      update_cache_variable(USE_CUDA OFF)
    endif (CUDA_FOUND)
  else (CMAKE_VERSION VERSION_LESS 3.17)
    find_package(CUDAToolkit QUIET)
    if (CUDAToolkit_FOUND)
      message("-- Found CUDA version ${CUDAToolkit_VERSION} installed in: "
        "${CUDAToolkit_TARGET_DIR}")
      set(CUDAToolkit_DEPENDENCY
        "find_dependency(CUDAToolkit ${CUDAToolkit_VERSION})")
      enable_language(CUDA)
      if (CMAKE_VERSION VERSION_GREATER_EQUAL "3.18")
        # Support Maxwell, Pascal, Volta, Turing, and Ampere GPUs.
        set(CMAKE_CUDA_ARCHITECTURES "50;60;70;80")
        message("-- Setting CUDA Architecture to ${CMAKE_CUDA_ARCHITECTURES}")
      endif()
      list(APPEND CERES_CUDA_LIBRARIES
        CUDA::cublas
        CUDA::cudart
        CUDA::cusolver
        CUDA::cusparse)
      set(CMAKE_CUDA_RUNTIME_LIBRARY NONE)
    else (CUDAToolkit_FOUND)
      message("-- Did not find CUDA, disabling CUDA support.")
      update_cache_variable(USE_CUDA OFF)
    endif (CUDAToolkit_FOUND)
  endif (CMAKE_VERSION VERSION_LESS 3.17)
endif (USE_CUDA)

MoMansa, 2024-05-31:
where did you copy the commend line in the Ceres library ?

mz7790, 2024-06-01:
> where did you copy the commend line in the Ceres library ?
He is located in CMakeLists.txt under the Ceres solver root directory.
As shown in the following figure：
[image]

MoMansa, 2024-06-03:
no sorry, what I want to know is in which file I have to put it