HKUST-Aerial-Robotics/A-LOAM #71: no point clouds in rviz

Addog666, 2024-12-05:
I followed the instruction and run the commands below:    

    roslaunch aloam_velodyne aloam_velodyne_VLP_16.launch
    rosbag play YOUR_DATASET_FOLDER/nsh_indoor_outdoor.bag

But there are no point clouds in the rviz, what should I do?

cuboya, 2024-12-06:
这样试试，后面加 --clock
rosbag play 2022-01-14-19-56-48.bag --clock

Addog666, 2024-12-06:
> 这样试试，后面加 --clock rosbag play 2022-01-14-19-56-48.bag --clock

加了--clock还是不行。我看这个bag发布的点云坐标系是/velodyne，但rviz里面的fix_frame是/camera_init，和这个有关系吗？我把rviz里面的fix_frame改成/velodyne，能看见bag发布的点云，但还是看不见建的地图

Addog666, 2024-12-06:
i found that there were no TF transformations from "/camera" to any other frames, should i add a TF transformation of "/camera" frame?

cuboya, 2024-12-06:
我这里的fix_frame就是/camera_init，刚才跑了一下可以出来建图

cuboya, 2024-12-06:
很奇怪的问题，你可以先看看riviz的topic的订阅有没有问题；当你运行别的slam算法生成地图时有没有点云呢

Addog666, 2024-12-09:
> 很奇怪的问题，你可以先看看riviz的topic的订阅有没有问题；当你运行别的slam算法生成地图时有没有点云呢

我解决了，可能是我用的ROS版本是noetic的问题，我把所有frame名字前面的“/”去掉了，然后就可以了

exDawn, 2024-12-09:
> > 我也没有，我只有轨迹没有点云，这里的状态信息说从 /camera_init 到 /camera_init 的坐标变换失败，我rqt查看tf树也是说No tf data received ![微信图片_20241209144130](https://private-user-images.githubusercontent.com/66551763/393729538-8a00e823-065b-443c-8b0e-39d7e2e4df01.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3MjkwMTQsIm5iZiI6MTczMzcyODcxNCwicGF0aCI6Ii82NjU1MTc2My8zOTM3Mjk1MzgtOGEwMGU4MjMtMDY1Yi00NDNjLThiMGUtMzlkN2UyZTRkZjAxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA5VDA3MTgzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE2M2FjNDg2MjliYzA4NjU3NDhiODM2YjE1Zjc3ZmU1NTc2MGQ4YjFlNzlkMTYwYmNmMzc2ZGI0ZjA1MjM2MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2VK0QveWj3uNvwxbxemZyZceSrbnoLDIGMco7oyqvAU)
> 
> 你用的是noetic版本的ROS吗？我把所有frame名字前面的“/”去掉了，然后就可以看见点云了

非常感谢，已解决